buildbotics/bbctrl-firmware #245: Manual firmware upload not working on MacOS

jcoffland, 2020-02-08:
Reported in #242.  Files are greyed out.

tmadrms, 2020-02-08:
Unable to upload firmware via any browser on MacOS -- firmware files are greyed out. 
[image]

jcoffland, 2020-02-09:
This is caused by a long standing OSX bug.  See https://bugs.chromium.org/p/chromium/issues/detail?id=521781 & https://bugzilla.mozilla.org/show_bug.cgi?id=1195508

Work around is to accept all ``.bz2`` files not just ``.tar.bz2``.

Should be fixed in [v0.4.13-rc20](https://buildbotics.com/bbctrl/bbctrl-0.4.13-rc20.tar.bz2).

tmadrms, 2020-02-09:
Confirmed working in v0.4.13-rc20.  Thanks!!

jcoffland, 2020-02-09:
Thank you for the report and confirmation.

closed 2020-02-09 by jcoffland