buildbotics/bbctrl-firmware #247: Ability to Have Separate Tools

leadgtr7, 2020-02-10:
I would like the ability to switch between my Huanyang VFD spindle using RS485 (with it's onw max and min spindle speed in my case 7800-24000) and my PWM laser engraver (speeds 0-255).

I don't mind to much just going into the tools and switching between them but it would be nice if each of the tools had their own settable parameters that would persist when changed

jcoffland, 2020-02-10:
You could handle this now by saving two different config.  So save a config go to the ``ADMIN -> General`` page and click ``Backup``.  Then restore the config you want to use from that same page.

leadgtr7, 2020-02-10:
Is that stored locally in the pi? It seemed like when I did that it was
stored in the machine that had the web browser connection

jcoffland, 2020-02-10:
It's not stored on the pi.  You would have to keep the files on another machine.