buildbotics/bbctrl-firmware #248: thumb drive corruption

DougCoffland, 2020-02-17:
If you format a thumb drive for FAT32 using Windows 10 and then simply plug it into the Buildbotics Controller, the drive becomes corrupted and causes a warning when you plug it back into Windows 10. When you plug it back into the Windows 10 machine the following warning appears in a window at the lower right corner of the screen.

"There's a problem with this drive. Scan the drive and fix it."

If you do nothing, the window goes away and the drive opens in Explorer, but continues to display the same message each time you plug it in. If you take their advice, the problem is cleared and the error message does not appear again until you plug it into the BB Controller again.

jcoffland, 2020-02-17:
This is because the drive is automatically mounted which marks the drive unclean.  If you eject the drive before pulling it out then the flag will be cleared.  The whole reason for this is that the OS may cache writes to the thumb drive so unless it is ejected the writes may not be complete.  Thus the drive may be in an inconsistent state.  One solution would be to mount thumb drives read-only.

DougCoffland, 2020-02-17:
When I try to eject the thumb drive on the BB Controller, I get a message that says:

Unable to unmount "usb-sda1"
umount: /media/usb-sda1: umount failed: Operation not permitted

DougCoffland, 2020-02-24:
I tried 0.4.14rc1 and things got really bad. When I plug in a thumb drive and attempt to load a file. The file dialog shows several instances of the thumb drive. When I remove the thumb drive, the instances are still there and eventually the browse hangs. They are still there after rebooting the controller.

I also tried ssh'ing to the controller and doing a mount command. The bogus instances of the thumb drive are still present in the browser on the local head, but mount does not report them.

Finally, when I 'Upgraded' back to 0.4.13, the bogus entries still show in the file selection dialog.

jcoffland, 2020-02-24:
What you are seeing are all the ``/etc/fstab`` entries.  I didn't realize they would show up even when unmounted.

jcoffland, 2020-02-26:
I've tried a number of different methods but I cannot find a way to automount USB drives in read/write mode and allow users to unmount it with out creating a pile of entries in ``/etc/fstab``.

One idea would be to automatically add an entry to fstab when the USB drive is plugged in and remove it when it's removed.

You will need to manually remove the offending entries from your ``/etc/fstab`` by editing the file.