buildbotics/bbctrl-firmware #249: Error when setting lower soft limit

DougCoffland, 2020-02-24:
When working from the local monitor, an error sometimes comes up when trying to set the lower soft limit. The error says:
Level: error
Source: Comm
Location: src/command.c:200
Repeat: 1
Message: Failed to parse float $0tn=-

Note, in this case I was trying to set the lower soft limit to -200 on Motor 0, which is the x axis. If I clear the error and refresh the page, the value is set back to 0. Also, this doesn't happen every time, but once it starts, I can get out of it by connecting across a network and then the value gets set correctly. When the error occurs, the value gets set to 0 even though it was previously set to -200.

tmadrms, 2020-02-24:
Thank you for opening this!  Sorry, I got distracted once I found the work around (setting it from remote computer).

jcoffland, 2020-02-24:
The error is saying that it tried the parse the value ``-`` which is not a valid number.

DougCoffland, 2021-03-04:
Another customer raised this issue again, so I did a bit more checking. He pointed out that the failure does not occur when using a remote connection.

I found that I can successfully set the soft limit on my local head when using the xontab soft keyboard with my ASUS VT168H touch screen monitor.

However, when I try to set the soft limit to a negative number using a hardware keyboard, I get the error code and failure every time.

DougCoffland, 2021-03-05:
fixed in V1.0

closed 2021-03-05 by DougCoffland