buildbotics/bbctrl-firmware #250: Oversized Max Values

tmadrms, 2020-02-24:
This may be a RTFM issue, but I'm starting to see a consistent issue where my sims local don't match machine max values.  For example the same file on local CAMotics vs Buildbotics controller sim.  This discrepancy means I cannot start my cut as it believes I'm going to go out of bounds.  You can see from the the two photos that the Buildbotics sim thinks the footprint is much larger than reality.  It seems to be adding a very large box around the final cuts.  This is even worse when using a final cut path (not shown).

[image]
[image]

jcoffland, 2020-02-24:
I think what is happening is that it's including an initial move from whatever position the machine was in when the simulation was run to the origin.  That's the slanted red line on the left.  This is extending the bounding box in both the X and Z axes.

tmadrms, 2020-02-24:
Machine was at x0y0 (work coordinates) at time of simulation. Keying off machine coordinates maybe?

tmadrms, 2020-02-24:
Figured out the culprit, but not sure why/how to repair.  I was using the max/min for the work area check bounding box (which was working prior to 0.4.13rc20) as part of the start script.  It seems it's now using that in the simulation and it takes the max/min values from Y of the machine, not the workpiece.  I just disabled them for now and things seem to work again.  Another great use for programmable buttons to alleviate this issue!