buildbotics/bbctrl-firmware #254: Evaluation of Named Global Variables Fail's in some/all circumstances

kwince, 2020-03-29:
In the file below, the move @line 53:
 G0    g0 x[#<_X_tool_len_pos>] y[#<_Y_tool_len_pos>]
Starts at 0,0,0 in the file.....and does not move, i.e., it moves to x0, y0 because the values are evaluated as '0'.

This is true in last two versions ending in xx.xx.12 and xx.xx.13. Version xx.xx.11 and previous do not have the problem.

;xy plane
G17 
;inch units
G20 
;Cutter compensation off
g40 
;use machine coordintates to startup 
g54
;absolute distance mode using current coordinate system
G90
;clear and disable global offset system
g92.1
;feeds in units per minute
G94 

m5

#<_Z_max>=7.9
#<_X_tool_len_pos>=0.50
#<_Y_tool_len_pos>=2.0

(msg, Place tool length stick with metal down and right. Keep pressed down, left, and against two parallel bars going left to right. When ready, click 'Continue')
	M0 ;pause for user input

;Z homing routine
  G28.2 Z0 
  F[#<_Z_search_velocity>]
  G38.6 Z[#<_Z_home_travel>]
  G38.8 Z[#<_Z_latch_backoff>] F[#<_Z_latch_velocity>]
  G38.6 Z[#<_Z_latch_backoff> * -8]
  G91 G0 G53 Z[#<_Z_zero_backoff>]
  G90 G28.3 Z0

;X homing routine 
  G28.2 X0 
  F[#<_X_search_velocity>]
  G38.6 X[#<_X_home_travel>]
  F[2]
  G38.8 X[#<_X_latch_backoff>]
  G38.6 X[#<_X_latch_backoff> * -8]
  G91 G0 G53 X[#<_X_zero_backoff>]
  G90 G28.3 X0

;Y homing routineS
  G28.2 Y0 
  F[#<_Y_search_velocity>]
  G38.6 Y[#<_Y_home_travel>]
  G38.8 Y[#<_Y_latch_backoff>] F[#<_Y_latch_velocity>]
  G38.6 Y[#<_Y_latch_backoff> * -8]
  G91 G0 G53 Y[#<_Y_zero_backoff>]
  G90 G28.3 Y0
  
   g0 x[#<_X_tool_len_pos>] y[#<_Y_tool_len_pos>]
	   (msg, check position1. When ready, click 'Continue')
	M0 ;pause for user in
	G0 z6.75
	G38.6 p18 z[#<_Z_max>-6.5] f3
	G0 x0 y0 z0
	
	#5000 = #5063 ; #5000 is taken in by notch and drill program
	m2

m2

DougCoffland, 2021-05-13:
It seems that named parameters that contain capital letters are not saved correctly. For instance, this program fails.
```
G21
#<Skipzprobing>=3
(DEBUG, parameter is #<Skipzprobing>)
M2
%
```
The resulting debug message is 'parameter is 0'.

However, this program works correctly.
```
G21
#<skipzprobing>=3
(DEBUG, parameter is #<skipzprobing>)
M2
%
```

I think the problem occurs when the parameter is assigned, because after assigning the parameter with no caps, it can be read back with caps in the name. However, if you assign the parameter with caps in the name, you cannot read it back correctly with no caps in the name.

According to the LinuxCNC documentation, named parameters are supposed to be case insensitive.

jcoffland, 2021-05-13:
v1.0.1-rc1 should fix this.

DougCoffland, 2021-05-13:
I tested v1.0.1-rc1 and it seems to be fixed. Thanks!

closed 2021-05-13 by DougCoffland

kwince, 2021-05-16:
Thank u for finding and fixing this.