buildbotics/bbctrl-firmware #258: long slow feeds lose axis distance reference

kwince, 2020-04-11:
When feeding at 7IPM at about 16 inches on my 36" X axis suddenly the display will go to negative. 14".

Version 0.4.1
I just boot up, zero the machine
zero it
go to the MDI
execute
g0 x0.5 y1.5
g0 z7.25
s17000
f7
g1 y1.25
; i may accientally type in g1 18 here instead of g1 x18, but I do follow it with g1 x18 and it starts to move
g1 x18
;at x=16 the display switches to x=-14 or so

kwince, 2020-04-12:
I will break it up and see if it works, probably will. I think this is some sort of overflow in math issue or a timeout issue. I will report about the divided up path and trying it on versions 0.4.12/13 versions. It would be nice if it's a timeout issue if the machine just stopped and the display flashed, or something less harmful than continuing with a negative value.

DougCoffland, 2020-05-25:
Another customer has experienced this problem. In this case, the user has a very slow feed rate causing a move that takes about 10 minutes. Here are the first few lines of his program:

G20 (inches)

M6 T1 (Change Tool: Diameter: 0.4375 in)

G0Z0.2500

G0X-5.7313Y-0.2188 (origin)

g0z-.63

G1X-5.2402Y-0.2188Z-0.6500F0.1 (origin 1)
G1X-5.7313Y-0.2188f4 (Origin 0)
G0X-5.2402Y-0.2188 (origin 1)

The line that reads G1X-5.2402Y-0.2188Z-0.6500F0.1 (origin 1) causes the move of about .5" on the X axis at a feed rate of .1 inches per minute. About half way through the move, the axis suddenly jumps and loses position. It happens every time.

kwince, 2020-05-26:
There is apparently a loop or move timeout, I believe.

I can figure it out if I split the move into a multiple moves and see what
happens. Won't be able to do that till next weekend though

mmcmilli, 2021-03-07:
Copied from my post on the BB Tech Forum:

"I was doing some fly cutting at 2IPM using MDI and noticed some unusual behavior. This behavior was repeatable from any location on the table and when run from a program. I only experimented in x axis, but this behavior may repeat for the others.
My program code was: "G01 G90 X-6.0 F2.0"
After travelling beyond -4.0" the machine would hick-up as if it lost steps and the position readout would flip sign to +4.**" It would continue feeding trying to get to -6" but would repeat the hickup and sign flip when it reached -4.**" again. It would continue this indefinitely. It also occured if I tried to go to +6" in my g-code but with all signs reversed.
After trying several things without affect, I found that if I changed the feed this fault would happen at different locations.
At 1IPM it was ~2"
At 2IPM it was ~4"
At 3IPM it was ~6"
At 4IPM it was ~8"
The pattern is approximately 2xfeed...on my machine.

I was able to work around this by breaking the long travel segment up into smaller chuncks or by using a feed that was higher. I think though, that long linear moves on larger machines could run into this problem at higher feeds based on the pattern here.

Things I tried that did NOT change the behavior.
Changing microstepping.
Changing Acceleration.
Changing Jerk.
Disabling limit switches.
Updating firmware from .4* to most current (New layout and features look awesome btw!)

I also didn't try any mult axis moves or arc moves to see if the problem occurred there. I mention it because others may be having this issue and need a work around and maybe it's something fixable in the code. My lead on the ballscrew would affect the distance travelled so I'll report that too (2.5mm).

Cheers!"

DougCoffland, 2021-03-08:
I have duplicated this problem. It happens exactly as described.

I simply set my travel per revolution on the X axis to 2.5 mm and ran the exact command that was described.

After it failed, I took looked at the log file and it appears to reverse mid-stream for no apparent reason. See below. Notice that it turns around at -110.952 mm, which is just about 4.37 inches.
**************************
I:Planner:set(#12, line, 1)
I:Planner:Cmd:{"max-vel": 50.8, "max-jerk": 150000000.0, "entry-vel": 0.0, "max-accel": 10000000.0, "type": "line", "first": true, "exit-vel": 0.0, "id": 13, "times": [34.92, 0.0, 34.92, 179930.17, 34.92, 0.0, 34.92], "target": {"x": -152.4}}
I:Comm:< #id=13\nlAAAAAAgJYYSwGA0PTQxZmYYww0Do4YOg2Do4YOg37uw/QA4Do4YOg6Do4YOg
I:Planner:Cmd:{"type": "end", "id": 14}
I:Comm:< #id=14\n
I:Comm:> {"0df":192,"xp":-0.079,"id":13,"v":0.051,"if":0.02,"xx":"RUNNING","xc":17}
I:Comm:> {"xp":-0.096}
I:Comm:> {"xp":-0.099}
I:Comm:> {"xp":-0.319}
I:Comm:> {"xp":-0.539}
I:Comm:> {"xp":-0.76}
I:Comm:> {"xp":-0.98}
I:Comm:> {"xp":-1.2}
I:Comm:> {"xp":-1.42}
I:Comm:> {"xp":-1.447}
I:Comm:> {"xp":-1.454}
I:Comm:> {"xp":-1.671}
I:Comm:> {"xp":-1.891}
I:Comm:> {"xp":-1.897}
I:Comm:> {"xp":-1.904}
I:Comm:> {"xp":-2.121}
I:Comm:> {"xp":-2.341}
I:Comm:> {"xp":-2.561}
I:Comm:> {"xp":-2.781}
I:Comm:> {"xp":-3.002}
I:Comm:> {"xp":-3.222}
I:Comm:> {"xp":-3.442}
I:Comm:> {"xp":-3.662}
I:Comm:> {"xp":-3.882}
I:Comm:> {"xp":-4.102}
I:Comm:> {"xp":-4.322}
I:Comm:> {"xp":-4.542}
I:Comm:> {"xp":-4.763}
I:Comm:> {"xp":-4.983}
I:Comm:> {"xp":-5.203}
I:Comm:> {"xp":-5.423}
I:Comm:> {"xp":-5.643}
I:Comm:> {"xp":-5.863}
I:Comm:> {"xp":-6.083}
I:Comm:> {"xp":-6.178}
I:Comm:> {"xp":-6.185}
I:Comm:> {"xp":-6.402}
I:Comm:> {"xp":-6.622}
I:Comm:> {"xp":-6.842}
I:Comm:> {"xp":-7.062}
I:Comm:> {"xp":-7.282}
I:Comm:> {"xp":-7.502}
I:Comm:> {"xp":-7.723}
I:Comm:> {"xp":-7.943}
I:Comm:> {"xp":-8.163}
I:Comm:> {"xp":-8.383}
I:Comm:> {"xp":-8.603}
I:Comm:> {"xp":-8.823}
I:Comm:> {"xp":-9.043}
I:Comm:> {"xp":-9.263}
I:Comm:> {"xp":-9.484}
I:Comm:> {"xp":-9.704}
I:Comm:> {"xp":-9.924}
I:Comm:> {"xp":-10.144}
I:Comm:> {"xp":-10.364}
I:Comm:> {"xp":-10.459}
I:Comm:> {"xp":-10.466}
I:Comm:> {"xp":-10.682}
I:Comm:> {"xp":-10.903}
[... 524 lines cut ...]
I:Comm:> {"xp":-110.945}
I:Comm:> {"xp":-110.952}
I:Comm:> {"0sl":false,"xp":110.78}
I:Comm:> {"0sl":true,"xp":110.56}
I:Comm:> {"xp":110.34}
I:Comm:> {"xp":110.327}
I:Comm:> {"xp":110.323}
I:Comm:> {"xp":110.103}
I:Comm:> {"xp":109.883}
I:Comm:> {"xp":109.663}
I:Comm:> {"xp":109.443}
I:Comm:> {"xp":109.222}
I:Comm:> {"xp":109.002}
I:Comm:> {"xp":108.782}
I:Comm:> {"xp":108.562}
I:Comm:> {"xp":108.342}
I:Comm:> {"xp":108.122}
I:Comm:> {"xp":107.902}
I:Comm:> {"xp":107.682}
I:Comm:> {"xp":107.461}
I:Comm:> {"xp":107.241}
I:Comm:I2C: S b=None w=None d=None
I:AVR:I2C: S b=None w=None d=None
I:Web:200 PUT /api/stop (10.0.0.17)
****************************

jcoffland, 2021-03-08:
I found and fixed this problem.  There was a numerical overflow.  You can try out the fix in the [v1.0.0-rc5 beta firmware](https://buildbotics.com/bbctrl/bbctrl-1.0.0-rc5.tar.bz2).  Note, this firmware release is not complete and has some other know problems.  However, these problems should be limited to the new web interface.

mmcmilli, 2021-03-08:
That's great!

DougCoffland, 2021-05-06:
Resolved in V1.0.0

closed 2021-05-06 by DougCoffland