buildbotics/bbctrl-firmware #272: rapid-auto-off remains active after changing spindle type

DougCoffland, 2020-06-07:
A user reported that the spindle was going off during rapid moves when using his Huanyang spindle. He had previously been using a laser. I suspected that 'rapid-auto-off' was still on, I configured my machine for a PWM spindle, checked 'rapid-auto-off' and saved the result. Then, I changed back to the 'Huanyang VFD', and saved it. Sure enough, 'rapid-auto-off' was still occurring.

I was able to work around this problem, by going back to 'PWM Spindle', unchecking 'rapid-auto-off' and saving the result. Then, after going back to 'Huanyang VFD' and saving, the problem was resolved.

I suspect that 'dynamic-power' may have the same problem. If so, this could potentially be more serious because it could cause the Huanyang spindle to turn slowly. If it is an air-cooled spindle, it could burn up.

jcoffland, 2020-12-30:
The fix is here: https://github.com/buildbotics/bbctrl-firmware/commit/83973f554fc07358326d334ca7157546e25dbf03.  It will be in the next release.

DougCoffland, 2021-05-06:
Fixed in V1.0.0

closed 2021-05-06 by DougCoffland