buildbotics/bbctrl-firmware #274: Automatic Tool Changer support

DougCoffland, 2020-08-12:
Add the ability to build a tool table. Also add the ability to automatically place the current tool back in the tool holder and fetch a new table from the tool holder. The tool table must document the tool length and the automatic tool changer must adjust the offset for the length of the tool.

Skyler1440, 2020-11-02:
Also, if the tool length offset parameters were persistant, that would be great. A graphical tool table like major competitors have would be a great addition.

cwallen93117, 2020-11-17:
Skyler - In the interim you can add the tool table to the program-start GCode.  This will cause it to be saved.  I have an excel spread sheet which calculates all my offsets and GCODE based on the measured Z and I can cut and paste it in depending on which Probe I'm using.    See below and attached.

(Fixed Probe Tool Table)
G10 L10 P1 Z-38.023 (Tool #1)
G10 L10 P2 Z-40.946 (Tool #2)
G10 L10 P3 Z-68.594 (Tool #3)
G10 L10 P4 Z-36.219 (Tool #4)
G10 L10 P5 Z-8.06 (Tool #5)
G10 L10 P6 Z-2.445 (Tool #6)
G10 L10 P7 Z-45.568 (Tool #7)
G10 L10 P8 Z-30.755 (Tool #8)
G10 L10 P9 Z-39.445 (Tool #9)
G10 L10 P10 Z-36.544 (Tool #10)
G10 L10 P11 Z-27.788 (Tool #11)
G10 L10 P12 Z-25.209 (Tool #12)
G10 L10 P13 Z-2.336 (Tool #13)
[Tool Table.xlsx](https://github.com/buildbotics/bbctrl-firmware/files/5554135/Tool.Table.xlsx)

cwallen93117, 2020-11-17:
Also you should run this at end to clear tool offsets
(Runs on M2, program end)

G0 X0 Y0 (go back to work coordinate Y and Y zero)
G92.1 (Clear Tool Offsets)
G92 X0 Y0 (Set work X and Y coordinates back to Zero in case I want to run program again with another piece of Stock)
M2 (Program End)