buildbotics/bbctrl-firmware #276: Jogging on Z axis does not stop

DougCoffland, 2020-09-17:
This issue occurs when jogging from the "Jog" tab on the "Control" page and with the "step" checkbox active. If you click the blue up arrow or the blue down arrow to jog the "Z" axis, it takes off and does not stop. This only seems to happen on the outermost ring (labelled 100). The inner rings seem to jog correctly.

This can be a real problem because you have to click the stop button to stop it before it runs into the top end stop going up or drives the bit into the table surface going down. Unfortunately, the stop button is not present on the "Jog" tab, so you have to change screens. By that time the crash has probably already occurred.

kwince, 2020-09-18:
I feel that the way the dog system works is that your input immediately
stores some amount of impulse into it that it has to dissipate with travel.
This would be count exactly opposite what I would want I would want the
movement to begin to accelerate and the minute you let go it would destroy
all that it had accumulated and do Max deceleration

DougCoffland, 2020-09-24:
Jogging in step mode moves specified distance, which is 100mm with fine adjust at max. I'm pretty sure this is what is causing the confusion.

closed 2020-09-24 by DougCoffland