buildbotics/bbctrl-firmware #277: Spindle, coolant, and mist come on when jogging from fog tab

DougCoffland, 2020-09-29:
If you start the spindle, mist, and/or coolant using M3 SXXXX, M7, or M8, then click the stop button. The spindle, mist, or coolant will start and then stop as expected.

However, if you go to the "Jog" tab in "Step" mode, the spindle, coolant, or mist will start again. Only the devices that were on when you clicked stop will restart. You have to go back to the MDI or Auto tab and click stop to get them to turn off.

You must be in "Step" mode for this to happen.

This happens every time. This also happens if you click the "Stop" button while running a GCode program that has the spindle, mist, or coolant active when you stop the program and attempt to jog in step mode from the "Jog" tab.

Issuing an M2 command will clear the problem.

It is a pretty serious problem because starting the spindle unexpectedly is a safety hazard.

jcoffland, 2020-09-29:
This is actually caused by running M72.  In step mode, the jog buttons execute something like this:

    M70
    G91
    G0X1
    M72

DougCoffland, 2020-10-07:
This is fixed in latest version of 0.4.15

closed 2020-10-07 by DougCoffland