buildbotics/bbctrl-firmware #284: G43 not working

Skyler1440, 2020-11-02:
For some reason, G43 doesn't do anything, rendering the tool offset codes dysfunctional.

jcoffland, 2020-11-04:
Can you provide an brief example?  Note, you have to select a tool before you can set its length offset.

Skyler1440, 2020-11-04:
For example:

G10 L1 P1 Z1.5 (set tool 1 Z offset from the machine origin to 1.5)
T1 M6 G43 (load tool 1 and tool length offsets, Z is at machine 0 and DRO shows Z1.500)

When I run this code, the position readouts don't change.

Thanks for your help!

jcoffland, 2020-11-11:
I see the problem. If you run this:

```
G10 L1 P1 Z1.5
T1 M6 G43
G0 X10
```

The incorrectly machine moves to (10, 0, 0).  However, if you run this:

```
G10 L1 P1 Z1.5
T1 M6 G43
G0 X10 Z1
```

The machine goes to the correct position (10, 0, 2.5).  The problem is that it's not seeing that the position has changed.