buildbotics/bbctrl-firmware #287: rapid-auto-off is honored by non PWM tool

BryanHunt, 2020-12-30:
I'm running a spindle as a custom VFD tool and I discovered that the speed was being set to 0 during a rapid.  I found that I had to go to the tool panel, select the PWM tool, uncheck the rapid auto off box, then switch back to custom VFD.  Doing this fixed the problem.

jcoffland, 2020-12-30:
That is intentional.

closed 2020-12-30 by jcoffland

BryanHunt, 2020-12-30:
And completely unintuitive.

jcoffland, 2020-12-30:
I looked at this again.  Now I see your complaint.  The problem is that when you switch to VFD mode, the option ``rapid-auto-off`` disappears but if left on it will still apply.  This issue is actually already recorded here: https://github.com/buildbotics/bbctrl-firmware/issues/272.  We plan to fix it soon.

BryanHunt, 2020-12-30:
Awesome.  Thank you!

jcoffland, 2020-12-30:
The fix is here: https://github.com/buildbotics/bbctrl-firmware/commit/83973f554fc07358326d334ca7157546e25dbf03.  It will be in the next release.