buildbotics/bbctrl-firmware #289: LinuxCNC-on-RPi

r13doc, 2021-02-03:
Does you controller work without wi-fi connection? 
Can I use ruspberry pi with a touchscreen function on  linuxCNC   with Buildbotics Controller ??

jcoffland, 2021-02-03:
The Buildbotics controller can be accessed with wifi or wired ethernet.  you can also connect a touch screen or mouse, keyboard and monitor directly to the controller.  LinuxCNC is not needed.  You can use [CAMotics](https://camotics.org/) to simulate your GCode and send it to the Buildbotics controller.

closed 2021-02-03 by jcoffland

kwince, 2021-02-04:
No. I have the controller hooked up to an ethernet cable, but I use my
phone on wifi to on the LOCAL NETWORK ONLY (cellular data off) to connect
to the controller.

r13doc, 2021-02-04:
Thanks, tell me please can I use this controller locally, for example
through camotics installed on my deb-ruspberry pi , without wifi or
ethernet ?
I am from Russia there is a problem with Ethernet and wi-fi...
Can I use for example nema 23 57HS76-3004 with this controller?

jcoffland, 2021-02-04:
You can use it with out Wifi or Ethernet if you put the GCode on an SDCard.   CAMotics will not run on the RPi.  You can use petty much any nema 23 motor as long as it's not uni-polar.  The one you listed will work.

r13doc, 2021-02-04:
Cool, thanks! You are the best!!

kwince, 2021-02-04:
Denis,
Gdyeh v'Russiya ti  zhivyosh?