buildbotics/bbctrl-firmware #290: Errors running "make camotics" during firmware build

MColeman0102, 2021-03-06:
Hi,
    I was following the development.md and I'm having trouble with "make camotics". Everything seems fine up until this point:
scons: `lib/libcbang-boost.a' is up to date.
scons: `lib/libyaml.a' is up to date.
build_function(["build/build_info.cpp"], [])
[Errno 2] No such file or directory
[Errno 2] No such file or directory
scons: `lib/libcbang.a' is up to date.
scons: done building targets.
+ export CBANG_HOME=/mnt/host/cbang
+ CBANG_HOME=/mnt/host/cbang
+ scons -C camotics build/camotics.so with_gui=0
scons: Entering directory `/mnt/host/camotics'
scons: Reading SConscript files ...
IOError: [Errno 2] No such file or directory: 'build/version.txt':
  File "/mnt/host/camotics/SConstruct", line 10:
    with open('build/version.txt', 'w') as f: f.write(version)
+ cleanup
+ umount /home/mark/bbctrl-firmware/rpi-root/dev/pts /home/mark/bbctrl-firmware/rpi-root/dev /home/mark/bbctrl-firmware/rpi-root/sys /home/mark/bbctrl-firmware/rpi-root/proc /home/mark/bbctrl-firmware/rpi-root/boot /home/mark/bbctrl-firmware/rpi-root/mnt/host /home/mark/bbctrl-firmware/rpi-root/

It looks like the camotics dev image might be missing the version.txt file for some reason.

Thanks!
Mark

jcoffland, 2021-03-06:
I think the problem is that the ``build`` directory is missing.  Try creating that first.

MColeman0102, 2021-03-09:
Hey,
   Yes the build directory was missing from the camotics_dev.img file. I mounted it, created the build folder and closed it. The build proceeds but now I'm receiving this:

**Exception: Chakra or V8 support is required, please rebuild C!. You may need to set CHAKRA_CORE_HOME or V8_HOME from /mnt/host/camotics/SConstruct line 85:**

I'll dig into that and see what I come up with.