buildbotics/bbctrl-firmware #291: 3D view not working on Chrome

DougCoffland, 2021-03-17:
The 3D View does not display on Chrome. I tried if from my Debian desktop with Chrome version 74.0.3729.108 and from a Mac laptop with Chrome version 89.0.4389.82.

It works with Firefox

jcoffland, 2021-03-17:
It works in Chrome for me. Chrome version 85.0.4183.102.  I doubt it's a Chrome issue.  Your browsers probably lack WebGL support.  Visit this site to check: https://get.webgl.org/

DougCoffland, 2021-03-17:
I went to that web site. It displays a rotating cube which suggests that I have webgl installed.

jcoffland, 2021-03-17:
What do you see at this page?:  https://webglreport.com/

DougCoffland, 2021-03-17:
It says "✓ This browser supports WebGL 1" at the top of the page.

DougCoffland, 2021-03-17:
It also says "✓ This browser supports WebGL 2" at the top of the second tab.

DougCoffland, 2021-03-18:
Fixed in 1.0 rc7. May require clearing cookies.

closed 2021-03-18 by DougCoffland