buildbotics/bbctrl-firmware #293: Can't run GCode program from Start Button

DougCoffland, 2021-03-24:
I am testing V1.0.0 RC8. I found that I am unable to run a GC0de program from the Start button. When I try to run the program, I get an error message with the following text:

message | Planner | Home/sdca_200_10x4_360_1.gc:6 | 1 | Change tool and attach probe
-- | -- | -- | -- | --

debug | API | RequestHandler.py:63 | 1 | Traceback (most recent call last):   
  File "/usr/local/lib/python3.5/dist-packages/tornado/web.py", line 1509, in _execute
     result = method(*self.path_args, **self.path_kwargs)
   File "/usr/local/lib/python3.5/dist-packages/bbctrl-1.0.0-py3.5.egg/bbctrl/APIHandler.py", line 46, in put
     self.put_ok(*args, **kwargs)
   File "/usr/local/lib/python3.5/dist-packages/bbctrl-1.0.0-py3.5.egg/bbctrl/Web.py", line 396, in put_ok
     def put_ok(self): self.get_ctrl().mach.start()
   File "/usr/local/lib/python3.5/dist-packages/bbctrl-1.0.0-py3.5.egg/bbctrl/Mach.py", line 303, in start
     self.planner.load(path) TypeError: load() missing 1 required positional argument: 'done'

error | API | RequestHandler.py:61 | 1 | load() missing 1 required positional argument: 'done'

If I assign the same program to a Macro button and run it from the Macro, the error does not occur.

DougCoffland, 2021-03-24:
I reverted to V1.0.0 RC6 and the problem does not occur.

jcoffland, 2021-03-25:
Should be fixed in rc9.

closed 2021-03-25 by jcoffland