buildbotics/bbctrl-firmware #315: Macro Button Feature Request

cwallen93117, 2021-11-16:
When you set up a macro button it automatically runs the "program-start", "tool-change" and "program end" gcode.  Would be great if for each macro there was three check box so you could decide which of these additional gcodes would be run for that macro.  For example if I'm running a probe routine I may not want to run the same start gcode or tool change gcode.  Would add a lot more flexibility to the macro buttons.

FirebirdDP, 2021-12-30:
You can implement this logic by using Parameters either as part of your program GCODE or a button that sets the parameter for you.

It would be nice if the macro button supports some on-off indicator.

Here is the code of my macro button which toggles the #<_autoprobe> parameter.

 (Check state of Probe 1=On 0=Off)
o102 if [#<_autoprobe> EQ 1]
  (MSG, Automatic Probe is disabled)
  #<_autoprobe> = 0
o102 elseif [#<_autoprobe> EQ 0]
  (MSG, Automatic Probe is enabled)
  #<_autoprobe> = 1
o102 else
  (MSG, Automatic Probe is enabled)
  #<_autoprobe> = 1
o102 endif

Tool Change Code which has conditional logic if the #<_autoprobe> =1 then run the tool change script.

(Check state of Probe 1=On 0=Off)
o102 if [#<_autoprobe> EQ 0]
 M0 (MSG, Automatic Probe is disabled)
o102 elseif [#<_autoprobe> EQ 1]
 M0 (MSG, Automatic Probe is enabled)
 (Runs on M6, tool change)
M70 (Save Modal State)
G21 (use millimeters for length units)
G90 (incremental distance mode)
G92.2 (turn off G92 offsets but keep parameters)
G0Z200 (Rapid Move)
G92.3 (Restore G92 Offsets)
M0 (MSG, Change tool, and attach probe)
F100 (Feed Rate)
G91 (incremental distance mode)
G38.2 Z-100 (probe)
G92 Z19.05 (Coordinate System Offset)
G0 Z25
M0 (MSG, Remove probe)
M72 (Restore Modal State)
o102 else
 M0 (MSG, Automatic Probe variable not set!!!)
o102 endif

cwallen93117, 2022-01-10:
Thanks for the suggestions.  I currently have that logic implement to skip the tool change gcode as I have a routine that runs there.  Would be great if M61 Q0 G43 worked so I could change to probe tool and apply offsets without triggering Tool change but the M61 isn't supported yet.