buildbotics/bbctrl-firmware #316: error handling G53 commands i.e. move in machine coordinates

cwallen93117, 2021-12-09:
It appears that when the buildbotics generates both the gcode warnings and 3d view it ignores any G53 Machine position moves, e.g. “G53 G0 Z150 “ and treats them as starting from the work coordinate origin e.g. “G0 Z150”.  See the image from the 3d viewer below.  The work coordinates were set at the blue/green/red origin.  

 
[image]

jcoffland, 2021-12-11:
Looks like it doesn't take the work coordinates into account.

cwallen93117, 2021-12-11:
Anyway this can be fixed?Charles917-359-4714Sent from my smartphone

jcoffland, 2021-12-11:
I'll take a look.

stromi65, 2021-12-21:
look at [https://github.com/buildbotics/bbctrl-firmware/issues/297] - same problem, was not fixed yet

DougCoffland, 2022-03-23:
fixed in version 1.0.2

closed 2022-03-23 by DougCoffland