buildbotics/bbctrl-firmware #334: External buttons using mappable I/0

Q1WP, 2022-08-25:
Since you have now created the mappable I/0, it would be great if additional external buttons could be added and linked to either a UI function, or a macro.

Currently, the only external button is the e-stop and if you are not directly in front of the touchscreen, even the most trivial issue requires pressing the e-stop which causes all kinds of issues.

I have pressed the stop or pause button numerous times and it has been great not losing the position info.

I have used arcade buttons (limit switches) many times and they are great when you might have dirty hands and don't want to touch the screen or use a mouse.

I think 3 options would be beneficial to link to UI buttons.  Run/Pause, Stop, and View.  The view button could toggle between the 3D view page and the control page.  I find myself switching back and forth while checking on progress many times on long cuts.  The run/pause and stop buttons are self explanatory.  All of the other UI buttons are needed during input and I don't see a great benefit of having remote buttons for them.

The other part of the mappable external buttons would be to have a few, maybe 1-5 external buttons that could trigger macros.  I have macros for homing, probing, setting origins, etc., and it would be very cool to have an external button to trigger these instead of having to walk over to my computer tower to trigger them.

I don't think it is needed to go "crazy" with this and allow numerous mappable buttons, but a few options might be good.

And I know the wireless keypads have addtional buttons, but i don't know if it is more difficult to map the gamepad as there are numerous versions with different button counts, or just target a simple I/O input pin.

I think I have about 5 pins available on the DB25 as I use external drives.

I have attached a photo of a CNC drill press table that automates the drilling of holes.  It is a python script that runs on a Raspberry PI and uses arcade buttons to link to UI software buttons to prevent messing up the touchscreen.

[image]

jcoffland, 2022-08-25:
Related to #286.

jcoffland, 2022-08-25:
This was one of the goals of the mappable I/O change.