buildbotics/bbctrl-firmware #340: DB15 pins wrong Control->I/O Pins tab

DougCoffland, 2022-11-29:
The following pins shown for the DB15 breakout box on the Control page I/O Pins tab are wrong:

- Pin 2 shows step 3, but should be step 0
- Pin 3 shows step 2, but should be step 1
- pin 4 shows step 1, but should be step 2
- pin 5 shows step 0, but should be step 3
- pin 7 shows n/c, but should be +3.3V
- pin 8 shows n/c, but should be spin0-10

DougCoffland, 2023-10-29:
Update the DB15 connector image to show the following changes in V2 software:

- pin 8 is no a 5V pin
- pin 9 is now a 0-10V output
- also see the corrections for the step pins in

DougCoffland, 2023-10-29:
Also update the following labels on the following pins to reflect changes made in V2 software and the V17 boards.

- pin i is another 5V pin
- pin 9 is now 0-10V outut
- pin 1 is  not longer a fault output and is a no-connect

DougCoffland, 2023-10-29:
When these corrections are done, I need to update the manual with the images on the Settings->Control->I/O tab with the new screenshots.

jcoffland, 2023-10-31:
Is this what it should look like?

<pre>
           +---------------+
   en -  9 |               | 8 - 0-10v
dir 0 - 10 |               | 7 - 5v
dir 1 - 11 |               | 6 - 5v
dir 2 - 12 |               | 5 - step 3
dir 3 - 13 |               | 4 - step 2
  gnd - 14 |               | 3 - step 1
  gnd - 15 |               | 2 - step 0
  gnd - 16 |               | 1 - n/c
           +---------------+
</pre>

DougCoffland, 2023-10-31:
Pin 16 is a no-connect. Here's how it should look.

[image: image.png]

closed 2023-11-03 by jcoffland