buildbotics/bbctrl-firmware #345: Remote network view not working.

Q1WP, 2024-03-20:
I recently upgraded to the Pi4 controller and the network service isn't loading properly on remote computers.

The old Pi3 controller loaded properly using Firefox/Chrome/Edge/Safari.

With my current setup, all of my connections are remote to save Pi4 resources (no keyboard, no mouse, no monitor, no webcam, wired ethernet).  

- My garage computer (Windows11 & Firefox) loads and runs the programs and shows the remaining time and shows the correct name of the currently running file.

- My design computer (Windows10 & Edge) does not show the remaining time and does not show the currently running file, only the last run file, that happens to be a macro file.

- My mobile (IOS & Safari) does not show the remaining time and does not show the currently running file, only the buildbotics.nc file that was deleted in the filesystem during initial setup, and isn't even shown in the controllers file browser.

These 3 screenshots were taken while the same program was running.

And all 3 of these used to be in sync and correct on the Pi3 controller.

Any ideas?

[image]

[image]

[image]

jcoffland, 2024-03-22:
Have you tried a hard reload of the page?  Usually CTRL-SHIFT-R.  It could be that the browser has cached some old code.

Q1WP, 2024-03-22:
I hadn't, but just did and no change, and I checked Edge/Chrome/Safari/Firefox.

jcoffland, 2024-03-22:
Does the ETA work correctly if you open the same GCode file on each browser?

jcoffland, 2024-03-22:
Doug and I discussed this.  We think the problem is that each browser has it's own "active file".  If the active file in the browser is different than the file running on the controller then the ETA will not show up.  We plan to change this in a future release so that there is only one "active file" and any connected browsers automatically switch files when the active file changes on the controller.  Hopefully that makes some sense.

Q1WP, 2024-03-22:
So, scratch that.  I just cleared all the browser histories on all devices and it does seem to show properly now.  I was thinking of your forced refresh thought.  Reloading, forced reloading didn't do it, but clearing all cookies did!  So I guess the issue for me is resolved, even though it might still need the active file change.  You have no idea how inconvenient it was not being able to see progress!  Thanks for the quick responses.

Q1WP, 2024-03-22:
If I clear the cookies, it does at least show text in the reminting field, but it isn't correct and it isn't of the currently running file.  This is probably as you suggested with the active file, but why did it change with the Pi4 version?

jcoffland, 2024-03-22:
We made a change to the software.  It will be fixed in the next release.

Thanks for the report.  It's important for us to know how you're using the software and what kind of problems you encounter.